dalance/sv-parser #10: Preprocessor issue with empty ifdef

Raffi, 2020-04-15:
Using "parse_sv -p" to parse the following code:
```
module A;
`ifdef OPT_1
  //wire a = 1'b1;
`else
  wire a = 1'b0;
`endif
`ifdef DEBUG
  `ifdef OPT_2
  //wire b = 1'b1;
  `else
  wire b = 1'b0;
  `endif
`endif
endmodule
```
results in the following output:
```
module A;
wire a = 1'b0;
wire b = 1'b0;
  endmodule
```
The line `wire b = 1'b0;` should not be there as DEBUG is not defined.

**Please note:** To see the issue we need both lines in OPT_1 and OPT_2 to be commented out as seen in the code above. When enabling the commented out lines, everything works as expected.

dalance, 2020-08-28:
Thank you for your information, and sorry for the late reply.
I fixed this issue.

closed 2020-09-02 by dalance

Raffi, 2020-09-03:
Thanks a lot for all the work you are doing, it is very much appreciated!