dalance/sv-parser #11: Accept trailing commas in module definition

tux3, 2020-05-21:
This syntax is accepted by many tools (including Yosys):

```
module m(
    input a,
    output b,
);
```

It'd be convenient to support it in sv-parser

dalance, 2020-05-28:
Thank you for your suggestion.
I like trailing commas and want that it is supported by SystemVerilog.
But unfortunately it is not in IEEE 1800-2017 and almost all commercial EDA tools don't support it.
So I think it should be detected as parse error.

tux3, 2020-05-29:
Yes, unfortunately it is not standard.
I have actually found that trailing commas in the module declaration is accepted by multiple commercial tools, unlike in the port connections where it is almost always rejected.
At least Aldec Active-HDL and Xilinx will accept it for synthesis (although Xilinx still highlights it like an error, all while accepting it...).

I can't really object to it being reported as a syntax error, it would only be convenient to accept existing code-bases. Please feel free to close this if you'd rather not add the option! =)

DaveMcEwan, 2022-06-10:
@tux3 Although it isn't a fix, I have found the comma-leading style helps for new code: https://github.com/DaveMcEwan/dmpvl/blob/master/prs/shr/sv/style_commaleading.sv