dalance/sv-parser #13: the concern to write sv-parser with RUST?

zhuzhzh, 2020-07-01:
RUST doesn't provide the tools likes flex or bison. So it needs lots of handy work.
what's the benefit to wirte the sv parser with RUST?

dalance, 2020-07-01:
I want to write SystemVerilog tools with Rust, So I wrote sv-parser with Rust.
I think it is difficult to handle the complicated data structure like syntax tree through FFI between C and Rust.
Additionally, I think flex/bison can't reduce handy work dramatically because the syntax of SystemVerilog is large and complicated.

zhuzhzh, 2020-07-02:
thanks for your answser! actually I find this sv-parser basically is the best sv front-end parser. but I am hesitating to learn RUST. I am think if you're intested to do some further work. for example, translate AST to RUST, then compile RUST into executable. so it could become one simulator.

dalance, 2020-07-02:
The implementation of simulator is too heavy work to me, and there are many competitors including commercial tools.
I'm interesting in more little tools to improve development like linter, formatter, refactoring support, and so on.
For example, there are some tools using sv-parser like below:

https://github.com/dalance/svlint
https://github.com/dalance/svls
https://github.com/sgherbst/svinst
https://github.com/zarubaf/morty

zhuzhzh, 2020-07-04:
understood, thanks!

closed 2020-07-04 by dalance