dalance/sv-parser #14: support macro in the files

zhuzhzh, 2020-07-02:
The file list is like blew shown:

define.sv
top.sv
undefine.sv

define.sv:
`define WIDTH 32

top.sv:
module test;
reg [`WIDTH-1:0] cnt;
reg clk=0;
always @(posedge clk) begin
    cnt <= cnt + 1;
end
end

undefine.sv
`undefine WIDTH

sv-parser can't support this style.

dalance, 2020-07-03:
I couldn't reproduce it. Could you explain more detailed information?
I tried like below:

```console
$ cat define.sv
`define WIDTH 32
$ cat top.sv
module test;
reg [`WIDTH-1:0] cnt;
reg clk=0;
always @(posedge clk) begin
cnt <= cnt + 1;
end
endmodule
$ cat undefine.sv
`undef WIDTH
$ parse_sv define.sv top.sv undefine.sv
parse succeeded: "define.sv"
parse succeeded: "top.sv"
parse succeeded: "undefine.sv"
```

zhuzhzh, 2020-07-04:
sorry, because I can't directly use sv-parser, I tried svinst. it has issue.

$ svinst  define1.v test1.sv define2.v dut.v 
files:
  - file_name: "define1.v"
    defs:
parse failed: "test1.sv" (Define not found: WIDTH)
  - file_name: "define2.v"
    defs:
parse failed: "dut.v" (Define not found: WIDTH)

zhuzhzh, 2020-07-04:
$ cat define1.v 
`define WIDTH 16

# harriszh @ hsv-sw02 in /lan/cva/ixcom_rt2/harriszh/tests/svinst [3:59:46] 
$ cat test1.sv 
moudle test;

reg [`WIDTH-1:0] cnt;
reg clk=0;
always #1 clk = ~clk;

dut u0(clk);

endmodule

# harriszh @ hsv-sw02 in /lan/cva/ixcom_rt2/harriszh/tests/svinst [3:59:53] 
$ cat define2.v 
`define WIDTH 32

# harriszh @ hsv-sw02 in /lan/cva/ixcom_rt2/harriszh/tests/svinst [3:59:58] 
$ cat dut.v 
module dut(input clk);

reg [`WIDTH-1:0] cnt;

always @(posedge clk) begin
	cnt <= cnt + 1;
end

endmodule

closed 2020-07-04 by zhuzhzh