dalance/sv-parser #17: Parse failure on parameter assignment

davidlenfesty, 2020-08-14:
I'm writing a utility based on this crate, and I've run into a bit of a snag with this bit of verilog:

```SystemVerilog
package new_pkg;
        parameter [48*8-1:0] NAME = {<<32{ {<<8{"MODULE NAME", {(48-11){8'h00}}}}}};
endpackage
```

(editorialized to protect the innocent, obviously, but that definitely fails parsing)

From what I can tell, this is valid SystemVerilog, and both Vivado and Verilator parse it without error.

dalance, 2020-08-18:
Thank you for the detailed information.
In my understanding, `streaming_concatenation` is `primary` not `constant_primary` like below:

```
primary ::= // from A.8.4
...
| streaming_concatenation
```

Therefore `streaming_concatenation` can be used as the RHS of `parameter_declaration` because it requires `constant_primary`.
For example, the following code can be parsed correctly.

```verilog
module a;
initial begin
    bit [48*8-1:0] NAME = {<<32{ {<<8{"MODULE NAME", {(48-11){8'h00}}}}}};
end
endmodule
```

BTW, Cadence tool seems to treat the original code as syntax error.

davidlenfesty, 2020-08-18:
Thank you for your time and know-how.

I see from other issues you don't want to deviate from the spec at all, so we'll be fixing this on our end.

closed 2020-08-18 by davidlenfesty