dalance/sv-parser #20: Instance name defined in a text macro

sgherbst, 2020-10-27:
Thanks again for the great work on this project.  I ran into an issue when using a text macro to define an instance name.  Considering these two files, I would expect the syntax trees to be the same after pre-processing:

**Example 1**
```verilog
`define MOD_INST u_mysubmod
module mymod;
mysubmod `MOD_INST (); 
endmodule
```

**Example 2** (add extra line at the beginning to get line numbers to match)
```verilog
module mymod;
mysubmod u_mysubmod (); 
endmodule
```

However, the syntax trees do not match, as show in these two attached files.  In particular, the syntax tree in **Example 1** does not include the two parentheses in the module instantiation.  This makes me think that **sv-parser** is treating the module instantiation as a macro call.
[example_1.txt](https://github.com/dalance/sv-parser/files/5448065/example_1.txt)
[example_2.txt](https://github.com/dalance/sv-parser/files/5448066/example_2.txt)

It does seem that the syntax is a bit ambiguous in this case, but both examples do appear to behave the same way when simulated by Icarus Verilog, at least.

By the way, I am using **sv-parser** 0.8.2.

dalance, 2020-11-06:
Thank you for you report.
I fixed it, and released v0.8.3.

sgherbst, 2020-11-10:
Awesome, thank you!  I confirmed that it is working on my end, and will close the issue.

closed 2020-11-10 by sgherbst