dalance/sv-parser #21: [Feature Request] Option to Disable Preprocessor:w

davidlenfesty, 2020-11-11:
I'm building a tool that operates on the source text, prior to any build processes, and need to have some way to bypass the preprocessed text. As far as I can tell there is no way to surface the length of the original string, prior to being preprocessed.

A specific example I have is this:

```verilog
case (foo)
    STATE1: foo <= bar[1];
    STATE2: foo <= bar[2]'
    `ifdef OPTIONAL_STATES
    OPTIONAL1: foo <= bar[3];
    OPTIONAL2: foo <= bar[4];
    `endif
endcase
```

Adding text to the second state fails *I believe* because extra whitespace from the ifdef effects the length of the last case's text. (i.e. `tree.get_str(case_node).len()` is longer than it would be without preprocessing).

Ideally there could be a way to disable preprocessing and simply include these statements as nodes in the syntax tree, however I believe there's still

davidlenfesty, 2020-11-11:
My apologies, I accidentally submitted this issue before fully qualifying it, and in doing so I realised how much of a change that would be and how preprocessor statements don't really fit into the tree (I believe every node would optionally need to contain a preprocessor directive which would be a nightmare from my understanding of the tree and the library).

I have workarounds for my specific issue, so this is not essential in my use case. Feel free to close this issue.

(I would still appreciate being able to do this, but I've realised I don't know if it's in scope for this library)

dalance, 2020-11-16:
In this case, you may be able to use `unwrap_locate!` and `SyntaxTree::get_origin` like below:

```rust
let locate = unwrap_locate!(case_node);
let origin = syntax_tree.get_origin(&locate);
```

`origin` is `Option<(&PathBuf, usize)>`, which means the filename and position of the `case_node`.
So the length of the case statement can be calculated from  the positions of `case_node`  and `endcase_node`.

I can add option to disable preprocess, but I think it will cause syntax error in many cases.

davidlenfesty, 2020-11-17:
Thanks for the response.

Yep, a solution like that is my workaround. I've realised that in any cases that this affects me and I would want to disable the preprocessor, it would cause a syntax error without some serious modifications to the parsing.

closed 2020-11-17 by davidlenfesty