dalance/sv-parser #29: Inline comment in preprocessor macro consumes next line

zarubaf, 2021-01-08:
The support for inline comments is great! However, I was observing that it breaks with our codebase when comment stripping is disabled. In particular, the inline comment would also consume the next line comment.

We do use `pragma`'s relatively often: 
```verilog
  // Assumptions and assertions
`ifndef VERILATOR
// pragma translate_off
.. 
// pragma translate_on
`endif
```

`sv-parser` would correctly remove the `VERILATOR` define (as it isn't set) but would also strip the `//pragma translate_on` from the next line. The pragma, therefore, becomes unbalanced and the result is an incorrect code block.

Do you see an easy chance to resolve this i.e., only strip comments until the end of the line? Thanks.

dalance, 2021-01-19:
Thanks!
I fixed it, and released v0.10.5.

closed 2021-02-03 by dalance

zarubaf, 2021-02-04:
That is awesome! Thanks, I can confirm that it is working on my end. Sorry for the delay, somehow the first message slipped through the cracks!

Thanks again!