dalance/sv-parser #31: parse error in "foreach" with two-dimentional array

x74ea6f, 2021-03-19:
Thank you for the useful tool.
I noticed that I am using svls in VS Code, so let me know.
I was getting a parse error in multiple foreach of a 2D array. Below is the sample code.

```
module example_foreach;

  int a[2][3] = '{'{1,2,3}, '{4,5,6}};

initial begin
    foreach(a[i]) foreach(a[i][j]) $display("%d", a[i][j]);
end

endmodule
```

dalance, 2021-03-19:
Thank you for your report.
In my understanding, `foreach` of a 2D array is not allowed in language specification like below.

```
 foreach ( ps_or_hierarchical_array_identifier [ loop_variables ] ) statement
```

I tried some commercial tools, so some tools can parse it, and some tools are failed to parse it.

x74ea6f, 2021-03-19:
Thank you for your reply.
I understand and close this.

closed 2021-03-19 by x74ea6f