dalance/sv-parser #44: Public access to identifier, text, and origin of processed Defines

niosHD, 2021-11-09:
I am currently trying to extend a SystemVerilog language server based on sv-parser with better define handling. In particular, I want to implement hover tool tips that show the value of a define and also want to support goto definition for defines.

For that, I need access to the value of a define and also the position where it has been defined. However, as far as I can tell, accessing the content of the Defines that are returned by the preprocessor is currently not supported because the struct members are not public. Is this a deliberate limitation or did I simply miss how it can be done?

Any hint is much appreciated. Thank you!

dalance, 2021-11-12:
Thank you for your interesting in sv-parser.
This is not an intended limitatin.
I added public access to these members, and released v0.11.2.

niosHD, 2021-11-14:
Great, works like a charm now.
Thank you for the fast release and for your work on sv-parser!

closed 2021-11-14 by niosHD