dalance/sv-parser #52: Question: Help getting started using parser.

prokie, 2022-04-13:
Hello.

I am trying to use this parser to find all ports declared in a module. This is the file I am using (very basic).

SystemVerilog file.
```
module adc(input rst_n);
endmodule
```

My match statement:
```
RefNode::AnsiPortDeclarationNet(x) => {
      println!("{:?}", x);
      let id = unwrap_node!(x, PortIdentifier).unwrap();
      let id = get_identifier(id).unwrap();
      let id = syntax_tree.get_str(&id).unwrap();
      println!("port_name: {}", id);
  }
```
So inside x, I have all the information I need about the port. The direction, type, and name. With the code, I can get the name of the port. But I don't know which identifier to use with `unwrap_node` to get the direction for example? 

When I print x I get: `AnsiPortDeclarationNet { nodes: (Some(NetPortHeader(NetPortHeader { nodes: (Some(Input(Keyword { nodes: (Locate { offset: 11, line: 1, len: 5 }, [Space(Locate { offset: 16, line: 1, len: 1 })]) })), DataType(NetPortTypeDataType { nodes: (None, ImplicitDataType(ImplicitDataType { nodes: (None, []) })) })) 
})), PortIdentifier { nodes: (SimpleIdentifier(SimpleIdentifier { nodes: (Locate { offset: 17, line: 1, len: 5 }, []) }),) }, [], None) }`

dalance, 2022-04-13:
There are some actual examples in [svlint](https://github.com/dalance/svlint) repository.
You can use `unwrap_node` repeatedlly.

https://github.com/dalance/svlint/blob/7379fdb1082e1ce026878c51778ac007e33b5e1c/src/rules/inout_with_tri.rs#L22

Or, you can decompose `nodes` like below:

https://github.com/dalance/svlint/blob/7379fdb1082e1ce026878c51778ac007e33b5e1c/src/rules/if_with_begin.rs#L23

prokie, 2022-04-13:
Thanks for the quick response. Will look into them! 👍

closed 2022-04-13 by prokie