dalance/sv-parser #54: Parse error for multiple imoprts

noah95, 2022-05-10:
When trying to parse the [`snitch` IP](https://github.com/pulp-platform/snitch/tree/master/hw/ip/snitch) I get a parse error, I guess due to the multiple `import` statements?

```
Error: parse error
   --> /usr/scratch2/lagrev1/huettern/git/snitch/hw/ip/snitch/src/snitch.sv:13:60
   |
13 | module snitch import snitch_pkg::*; import riscv_instr::*; #(
   |                                                            ^
Error: Parse error: Some(("/usr/scratch2/lagrev1/huettern/git/snitch/hw/ip/snitch/src/snitch.sv", 490))
```

### Steps to reproduce

- **sv-parser**: 0.11.1

```
git clone --depth=1 git@github.com:pulp-platform/snitch.git
cd snitch/hw/ip/snitch
bender sources -f > sources.tcl
morty -f sources.tcl
```

noah95, 2022-05-10:
**CC**: @zarubaf

DaveMcEwan, 2022-06-10:
I think sv-parser does the correct thing here. The correct syntax is `module snitch import snitch_pkg::*, riscv_instr::*; #(`.

In the LRM (IEEE1800-2017), page 697 shows an except from Annex A, where `module_ansi_header` permits a single optional `package_import_declaration`. The definition of `package_import_declaration` shown on page 777 allows multiple comma-separated `package_import_item`.

zarubaf, 2022-06-10:
@DaveMcEwan  Indeed I agree after looking into the Language Manual. Interesting, that most of the tools have no problem with the `;` though.

p. 781 also gives an example:

```verilog
module M import A::instruction_t, B::*; #(WIDTH = 32)
...
(input [WIDTH-1:0] data, input instruction_t a, output [WIDTH-1:0] result, output boolean_t OK
); endmodule: M
```