dalance/sv-parser #59: Parse fails for assign statement with intra-delays

bensampson5, 2022-07-14:
For the following module, the parser is unable to parse this. In your example from the README, it just outputs "Parse failed". I'm running version 0.12.0. It looks like it fails when using assign but is ok with always_comb. This is legal syntax as far as I can tell from IEEE Std 1800-2017. It's discussed in 9.4.5 on Intra-assignment timing controls

 ```systemverilog
`define DLY #10ps

module wire_with_delay (
    input logic a,
    output logic y
);

    assign y = #1 a;  // fails
    assign y = `DLY a;  // fails
    always_comb y = `DLY a;  // works

endmodule
```

dalance, 2022-07-14:
Section 9.4.5 dosen't define about `assign` statement. It defines blocking_assignment and nonblocking_assignment in `always`, `initial`, etc.

`assign` statement is defined at "10.3 Continuous assignments" in IEEE Std 1800-2017.
In this section, `assign` with delay is defined like below:

```
assign [ delay_control ] list_of_variable_assignments ;
```

So the following code is valid.

```verilog
assign #1 y = a;
```

bensampson5, 2022-07-14:
Ah yes - I stand corrected. Thanks for clarifying. So yes that is invalid syntax.

closed 2022-07-14 by bensampson5