dalance/sv-parser #66: Incorrect syntax error on generate begin-end block

MahmoudKMaarouf, 2022-10-27:
Having a begin-end block around a generate block causes an error. This is very difficult to avoid because it is common to name begin-end blocks for readability. 

Any help is appreciated.
```
generate 
         begin
     // begin: with_name_no_work_too
         genvar i;
         for (i = 1; i <= 4; i = i + 1)
           begin 
              temp[i]= 1;
           end
      end
   endgenerate
```

Error:
     |
18 |        begin: with_name_no_work_too
     |                ^

dalance, 2022-10-27:
`generate begin` is invalid syntax.
According to IEEE Std 1800-2017 A.4.2:

```
generate_region ::=
generate { generate_item } endgenerate
generate_item ::=
module_or_generate_item
| interface_or_generate_item
| checker_or_generate_item
```

`module_or_generate_item`, `interface_or_generate_item` and `checker_or_generate_item` can't have `begin/end`.

You can write like below instead of it.

```
generate
    if (1) begin
        genvar i;
        for (i = 1; i <= 4; i = i + 1)
        begin
            temp[i]= 1;
        end
    end

    if (1) begin: with_name_no_work_too
        genvar i;
        for (i = 1; i <= 4; i = i + 1)
        begin
            temp[i]= 1;
        end
    end
endgenerate
```

MahmoudKMaarouf, 2022-10-28:
Thank you Dalance!

closed 2022-10-28 by MahmoudKMaarouf