dalance/sv-parser #68: Syntax issues not being caught

MahmoudKMaarouf, 2022-11-05:
Some syntax issues are not being caught by this parser.
e.g.
**Data declaration outside a module:**
 logic a_var;
**this line inside a module:**
some_var;
**an incorrect data type:**
wir a_wire;

This tool can not be used as a syntax linter with these errors. Any suggests are appreciated.

DaveMcEwan, 2022-11-08:
@MahmoudKMaarouf Those examples are all valid syntax for a `package_item`, described at over pages 1136 and 1137 of IEEE1800-2017.
I *think* the intention is to enable interactive REPL interpreters, e.g. so you can experiment with expressions, but I can't say for sure.
This language feature is present in all versions of SystemVerilog (IEEE1800-2005,2009,2012,2017), but not in Verilog (IEEE1364-1995,2001,2005).

MahmoudKMaarouf, 2022-11-08:
> @MahmoudKMaarouf Those examples are all valid syntax for a `package_item`, described at over pages 1136 and 1137 of IEEE1800-2017. I _think_ the intention is to enable interactive REPL interpreters, [...]

@DaveMcEwan Yes but note that these lines were written for a module, not a package_item. Note that these do not compile as I have tested it myself as well. 

It is normal for a parser to ignore these errors, but, Dalance's svlinter here: https://github.com/dalance/svlint does not catch these errors as well, even though it is a linter, not a parser

DaveMcEwan, 2022-11-08:
These are not errors, so there is nothing for the parser to ignore.
`package_item` is a BNF clause, and is perfectly valid outside a module or package.
This is not a parser issue.

I think you're asking for a way to detect a (valid) language feature that you don't want to use.
For that, you can write a rule for svlint.
Indeed, that's what most of the functional rules are for, e.g. **keyword_forbidden_wire_reg** detects instances of the "wire" variant of the `net_type` BNF clause, and the "reg" variant of the `integer_vector_type` clause.
In your case, you probably want to detect instances of the `package_item` BNF clause, but only when not under a `package_declaration` instance.

MahmoudKMaarouf, 2022-11-08:
> These are not errors, so there is nothing for the parser to ignore. `package_item` is a BNF clause, and is perfectly valid outside a module or package. This is not a parser issue.
>
> I think you're [...]

Ah, I understand now. Essentially, the package_item enables custom defined types meaning it can be syntactically correct. Thank you!