dalance/sv-parser #72: `IFDEFs and such cause incorrect line numbers

MahmoudKMaarouf, 2022-11-19:
Placing this on top of code:
```
`ifndef AN_IF_DEF
logic [1:0] someVar;
`endif
```
will cause the line numbers of the original file to be shifted 1 up (equivalent to minus 1) more than what it currently is. More lines within the IF_DEF that are not executed will shift it even further, making the lines almost completely useless with large Verilog files and many IF DEFs. This is very hard to work around since the line numbers are critical for my tool.

Any help is appreciated.

DaveMcEwan, 2022-11-19:
@MahmoudKMaarouf Would you add a testcase to `sv-parser-pp/src/preprocess.rs`?
There are similar testcases called `ifdef_undefined` and `include_noindent`.
The important part is the check on, for example, `ret.origin(5).unwrap.l` (for the 5th character).
If you're not also fixing the bug (it might be difficult), you can add an attribute like `#[ignore = "Exposes issue #72, wrong line number."]`

https://github.com/dalance/sv-parser/blob/master/sv-parser-pp/src/preprocess.rs#L1213

MahmoudKMaarouf, 2022-11-19:
@DaveMcEwan 
Hi Dave, thank you for the suggestions.
I will think about adding these testcases, given that I have time.
May I ask how I run these test cases?

DaveMcEwan, 2022-11-21:
Nice and easy :) `cd sv-parser-pp/ && cargo test`

MahmoudKMaarouf, 2022-11-21:
Much appreciated Dave.
In the coming weeks, I will put some effort into this, and hopefully come up with a fix, or at least a testcase.