dalance/sv-parser #86: Bug: sv-parser doesn't handle sources which are symlink.

Graian, 2023-07-21:
While building rust source with the `rules_rust`(`rust_library` or `rust_binary`) build rule in the bazel build system, I encountered a bug. Bazel creates working directories using symlinks to files, but `sv-parser` checks if it's a file and ignores it if it's a symlink. so, requires an update to allow `sv-parser` to handle symlinks.

This line cause this issue.
- https://github.com/dalance/sv-parser/blob/master/sv-parser-syntaxtree/build.rs#L68

Graian, 2023-07-22:
When I modified the above-mentioned line of code to the following, I found that it fixed the problem. I don't know what side-effects this fix has on the overall behaviour of `sv-parser`.

`if entry.file_type().is_file()` ->` if entry.path().is_file()`

DaveMcEwan, 2023-07-24:
@Graian That modification looks sensible to me. There shouldn't be any unwanted side-effects.
Perhaps you could open a [PR](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request) so that it's easy to merge?

I'd like to understand more about the issue because it likely affects the related project svlint ([here](https://github.com/dalance/svlint/blob/master/build.rs#L407) and [here](https://github.com/dalance/svlint/blob/master/build.rs#L426)).
1. What are you using Bazel for? Instead of Cargo? Apologies if this is a silly question!
2. How can I (or others) reproduce your issue?
3. Is Bazel compatibility important enough to add [CI](https://docs.github.com/en/actions) [checks](https://github.com/dalance/sv-parser/blob/master/.github/workflows/regression.yml)?

Graian, 2023-07-25:
@DaveMcEwan , Thanks for your comment. I'll open a PR for this issue and here are my answers for your questions.

1. What are you using Bazel for? Instead of Cargo? Apologies if this is a silly question!
  > In my development environment, I'm using bazel as my build system, so I'm trying to build an existing Cargo project on top of bazel. The basic external dependency is on `Cargo.toml`, and bazel leverages `Cargo.toml` and the `Cargo.lock` file.

2. How can I (or others) reproduce your issue?
  > You can reproduce this issue with my test repository: [https://github.com/Graian/bazel_test/tree/main/sv_parser)].
  > You can refer bazel run command: [https://github.com/Graian/bazel_test/tree/main/sv_parser#bazel-run-command]

3. Is Bazel compatibility important enough to add [CI](https://docs.github.com/en/actions) [checks](https://github.com/dalance/sv-parser/blob/master/.github/workflows/regression.yml)?
  > As far as I know, I haven't had any major problems with it other than the symlink issue. However, It would be nice to see it added to CI checks.

Graian, 2023-07-27:
@DaveMcEwan , I've forked this repository and create a merge request!

- https://github.com/dalance/sv-parser/pull/87

DaveMcEwan, 2024-01-17:
@Graian Would you mind closing this issue as it looks like your fix is merged.