dalance/sv-parser #88: Can not parse `substr` method

Lockedloop, 2023-07-28:
I am using the `substr` method to decode multiple parameter values from one string parameter.
Svlint throws me a parse error:

```
Error: parse error
   --> rtl/my_file.sv:65:44
   |
65 | localparam int P1 = MY_STRING_PARAM.substr(1, 3).atoi();
                                               ^
```

DaveMcEwan, 2023-07-28:
I don't have the LRM on hand right now, but I think that is invalid syntax. Localparams must be assigned with a constant expression (fixed at elaboration), but `substr` is only evaluated at simulation (I think).
To convert to valid syntax, you can just remove the `localparam` keyword.

Lockedloop, 2023-07-28:
Hi, thank you for the prompt reply. 
Simulation works fine, also verilator lint passes on this file.
I removed the `localparam`, so I have `int P1 = ...` and the lint passes indeed.
However then the sim gets broken, because `P1` is no longer a constant expression.

DaveMcEwan, 2024-01-17:
@dalance I think this can be safely closed.

closed 2024-01-17 by dalance