dalance/sv-parser #89: can't parse when resetall is present and module has multiple parameters

charletes, 2023-08-11:
Hi, I'm trying to parse the following code and it fails when reaching the `#`:

```verilog
`resetall
`timescale 1ns/1ps

module test #(
    parameter VAL1 = 3,
    parameter VAL = 4
) (data_out);

output data_out;

endmodule
```

However, this works:

```verilog
// `resetall <------------
`timescale 1ns/1ps

module test #(
    parameter VAL1 = 3,
    parameter VAL = 4
) (data_out);

output data_out;

endmodule
```

This one also works:

```verilog
`resetall
`timescale 1ns/1ps

module test #(
//    parameter VAL1 = 3, <------------
    parameter VAL = 4
) (data_out);

output data_out;

endmodule
```

Any idea about how to proceed?

charletes, 2023-08-11:
This is how I'm processing it btw:

```rust
        let mut defines = HashMap::new();
        let includes = Vec<PathBuf>::new();
        let res = preprocess(&path, &defines, &includes, false, false);

        match res {
            Ok((text, new_defines)) => {
                println!("Yay!");

                let orig = String::from(text.text());
                let res = parse_sv_pp(text, new_defines, false);

                match res {
                    Ok((syntax, new_defines)) => {
                        println!("yay again!");
                    }
                    Err(x) => { // <- this is the error I get
                        println!("booo (2)");
                        debug_println!("{:?}", x);
                        print!("{}", orig);
                    }
                }
            }
            Err(err) => {
                println!("Boo!");
            }
        }
```

The output is:

```
Yay!
booo (2)
Parse(Some(("test_module.sv", 45)))
```

DaveMcEwan, 2023-08-14:
I can reproduce this with svlint v0.9.0.
```systemverilog
`ifdef DIRECTIVE_RESETALL
`resetall
`endif// DIRECTIVE_RESETALL
module M
#(
P
`ifdef SECOND_PARAM
,
`ifdef KEYWORD_PARAMETER
parameter
`endif// KEYWORD_PARAMETER
Q
`endif// SECOND_PARAM
) () ; endmodule
```

Omitting any of the preprocessor defines avoids the parse error. I must define all 3 before I see the parse error:
`svlint-parseonly foo.sv -D DIRECTIVE_RESETALL -D SECOND_PARAM -D KEYWORD_PARAMETER` ->
```
Error: parse error
   --> foo.sv:5:1
  |
5 | #(
  | ^
```

Output from the preprocessor looks good:
`svlint-parseonly foo.sv -D DIRECTIVE_RESETALL -D SECOND_PARAM -D KEYWORD_PARAMETER -E` ->
```
`resetall
// DIRECTIVE_RESETALL
module M
#(
P
,
parameter
// KEYWORD_PARAMETER
Q
// SECOND_PARAM
) () ; endmodule
```

Sorry @charletes, I don't know *why* this happens, but hopefully it's good to know that it's not just you.

charletes, 2023-08-14:
Thanks for looking into it! I'll try to catch where the tool breaks down but I'm not familiar with the code so it may take a while.

charletes, 2023-09-06:
Unfortunately I don't understand enough of the internals to figure out where things go awry.  I had a look at the parsing process and it seems to follow the syntax definition to a tee, but somehow the "resetall" gets things in the wrong state.

I did a really ugly workaround that works for me, using the preprocessor first, removing all the resetall statements, and then parsing the resulting string.

yousifBilal, 2024-03-26:
Any progress on this? I also encountered an issue with an internal design using `resetall macro. But the error position points to random things.