dalance/sv-parser #94: Comment in macro causes syntax error

skjdbg, 2024-01-15:
If we use a comment on the last line of a macro such as the `// Bad` in the following code snippet, the code is poorly preprocessed which leads to a syntax error:

```systemverilog
`define foo(code) if (1'b1) begin code end else $display("Bar");

`foo(
   ; // Good
   ; // Bad
)
```

To my understanding, after the preprocessing step, the previous code should look like this:

```systemverilog
if (1'b1) begin 
   ; // Good
   ; // Bad
end else $display("Bar")
```

However the newline character of the line `   ; // Bad` is trimmed which leads to the following resulting code (if we don't strip the comments):

```systemverilog
if (1'b1) begin 
   ; // Good
   ; // Bad end else $display("Bar")
```

Which is a syntax error since the `end` keyword (and the rest of the code) is commented.

The faulty code seems to be a `trim_end()` call in `preprocess.rs`:
https://github.com/dalance/sv-parser/blob/f5b3f2b090aec7a02984505c5dd39da2d5d33594/sv-parser-pp/src/preprocess.rs#L935

Removing it solves my issue however it was added in a specific commit to fix a spacing issue (56a13e4) and since I don't have an example to work with, I don't know how to fix both.

DaveMcEwan, 2024-01-16:
Good question! There's also this at the bottom of page 676:
> If a one-line comment (that is, a comment specified with the characters //) is included in the text, then the comment shall not become part of the substituted text.

I'm not sure there is a correct answer because, in my opinion, the LRM is not precise enough.
To solve your syntax-error issue, you could use C-style comments like `/* ... */` instead.

The call to `.trim_end()` solves the issue of extra spaces when you call a macro like this.
```
`foo(good, bad   )
              ^^^
```
While you definitely want to remove most trailing whitespace (space, tab, LF, CR, line break, non-breaking space, and a few more), perhaps you could trim all except CR and LF? Digging through the source for `str::trim_end` -> `char::is_whitespace`, you could copy+modify [`is_whitespace`](https://doc.rust-lang.org/src/core/char/methods.rs.html#862) to avoid `x0a` and `x0d`... but I don't know if this will cause other formatting issues.

skjdbg, 2024-01-24:
I replaced the `trim_end` call by
```systemverilog
trim_end_matches(|c: char| c.is_whitespace() && c != '\x0a' && c != '\x0d')
```
And it does solve my issue while fixing most failing tests that simply removing the call triggers.

There remains the `macro_arguments` test where 
```systemverilog
`define disp(clk, exp, msg)\
  always @(posedge clk)\
    if (exp) begin\
      $display msg;\
    end\

module M ();

`disp(
  clk,
  !(a[i].b && c[i]),
  ("xxx(()[]]{}}}", a[i].b, c[i])
); // NOTE: Semi-colon is unnecessary.
```

is transformed into 
```systemverilog
`define disp(clk, exp, msg)\
  always @(posedge clk)\
    if (exp) begin\
      $display msg;\
    end\

module M ();

  always @(posedge clk)
    if (!(a[i].b && c[i])) begin
      $display ("xxx(()[]]{}}}", a[i].b, c[i])
;
    end
; // NOTE: Semi-colon is unnecessary.
```
instead of what is expected (notice the semicolon which is not on a newline)
```systemverilog
`define disp(clk, exp, msg)\
  always @(posedge clk)\
    if (exp) begin\
      $display msg;\
    end\

module M ();

  always @(posedge clk)
    if (!(a[i].b && c[i])) begin
      $display ("xxx(()[]]{}}}", a[i].b, c[i]);
    end
; // NOTE: Semi-colon is unnecessary.
```
In both cases the resulting code is valid, although not pretty in the first case.
I think that the first case could also cause some unintuitive indentation/formatting rule issues in svlint (well formated code becoming poorly formatted code after pre-processor) whereas simply using a C-style comment is not a big issue.

If you agree, I'll simply close this issue.

closed 2024-02-01 by skjdbg