dalance/sv-parser #97: thread 'main' has overflowed its stack

mlr11, 2024-05-09:
The piece of verilog code below is causing the parser to exit with:
> thread 'main' has overflowed its stack.  

I tried increasing the stack up to 64M without any luck.

The somewhat convoluted expression is giving a tough time to the parser.
```
module test (
    input wire clk,
    input wire reset,
    input wire lcomreq,
    input wire lmask1,
    input wire lmask0,
    input wire req3,
    input wire req2,
    input wire req1,
    input wire req0,
    output reg lgnt0
);

always @(posedge clk or posedge reset) begin
    if (reset) begin
        lgnt0 <= 1'b0;
    end else begin
        lgnt0 <= (~lcomreq & ~lmask1 & ~lmask0 & ~req3 & ~req2 & ~req1 & req0)
               | (~lcomreq & ~lmask1 &  lmask0 & ~req3 & ~req2 &  req0)
               | (~lcomreq &  lmask1 & ~lmask0 & ~req3 &  req0)
               | (~lcomreq &  lmask1 &  lmask0 & req0  )
               | ( lcomreq & lgnt0 );
    end
end

endmodule
```
Everything works as expected if I refactor the code and create intermediate terms for each part of the 'or'
The always block would look like:
```
always @(posedge clk or posedge reset) begin
    if (reset) begin
        lgnt0 <= 1'b0;
    end else begin
        lgnt0 <= term1 | term2 | term3 | term4 | term5;
    end
end
```
with wire assignment for each of the terms.

mlr11, 2024-05-27:
sv_parser might not be the culprit here.  I kept getting this error for the majority of the verilog designs I was trying to use.
Even some of the passing examples from the sv-test suite were failing with the same message.

I tried switching from Windows to Linux (WSL - Ubuntu 22.04) and all of my problems went away.

yousifBilal, 2024-08-14:
I also ran into stack overflows on some designs