dalance/sv-parser #98: compiler directive not showing in syntax_tree

mlr11, 2024-06-26:
The following single line verilog file:

``` `timescale 1 ns / 1 ps ```

produce the following syntax tree:

``` 
SourceText
   UnsignedNumber
    Token: '1' @ line:1
   TimeUnit
    Keyword
     Token: 'ns' @ line:1
   Symbol
    Token: '/' @ line:1
   UnsignedNumber
    Token: '1' @ line:1
   TimeUnit
    Keyword
     Token: 'ps' @ line:1
```

All the information that is part of the timescale is present but there's no information that this belongs to a timescale.

I noticed something similar with `undef MACRONAME

The name of the macro shows as a TextMacroIdentifier but there's no information about the undef part.

I haven't looked yet at how other directives are behaving (non macro or include file related).

yousifBilal, 2024-07-07:
There is a preprocessor that expands all macros. I have opened an issue about it [here](https://github.com/dalance/sv-parser/issues/96), hoping to get the syntax tree without preprocessing it.

mlr11, 2024-07-24:
> There is a preprocessor that expands all macros. I have opened an issue about it [here](https://github.com/dalance/sv-parser/issues/96), hoping to get the syntax tree without preprocessing it.

What I am looking for is a bit different.  I am looking for the information about the directives that are not handled by the preprocessor.  Things like `timescale, default_nettype, celldefine, endcelldefine, ...

I ended up finding them in the parseTree but they were attached to WhiteSpace, so I had to add processing for WhiteSpace in my traversal to extract the directives