dalance/sv-parser #101: parameter array of strings causes an syntax error

fpgauserdude, 2024-09-06:
```
localparam string TESTED_VERSIONS[] =
   {
     "2023.1",
     "2023.2"
    };
```

$ svinst test.sv
files:
parse failed: "test.sv"
 test.sv:72:42
   |
72 |   localparam string TESTED_VERSIONS[] =
   |                                                                ^

The syntax is maybe a little weird, but perfectly legal. This code compiles on a commercial tool.

dalance, 2024-09-13:
According to Formal syntax of LRM, `localparam` can take `unpacked_dimension` only, so `[]` can't be used.

```
local_parameter_declaration ::= localparam data_type_or_implicit list_of_param_assignments
list_of_param_assignments ::= param_assignment { , param_assignment }
param_assignment ::= parameter_identifier { unpacked_dimension } [ = constant_param_expression ]
unpacked_dimension ::= [ constant_range ] | [ constant_expression ]
```

dalance, 2024-09-13:
For example, Synopsys Formality outputs syntax error to this code.
A purpose of sv-parser is detecting such corner cases (almost all tools work fine, but some tools don't) by implementing as Formal syntax is.

fpgauserdude, 2024-09-13:
I can appreciate the value of finding code that is not strictly compliant with the LRM. Sometimes I think that HDL tool vendors do not do the developer community any favors by allowing code that is non-compliant.